QuirkyCort/gears #126: Football arena example

vsalcaidesfaria, 2023-10-20:
Hi!

Its possible someone  post an example to use the football arena and a simple code using the electromagnet sensors?

Thank you.

sgmurray, 2023-10-20:
Yes. There is example code here:

https://raw.githubusercontent.com/QuirkyCort/gears-contributions/main/code/python/Steven%20Murray%20-%20Football/football_sample.py

vsalcaidesfaria, 2023-10-23:
> Yes. There is example code here:
> 
> https://raw.githubusercontent.com/QuirkyCort/gears-contributions/main/code/python/Steven%20Murray%20-%20Football/football_sample.py

Thank you very much!

I will try the code. I'm preparing a school contest using the sumo and football arenas. 

Gearsbot is an amazing tool for this kind of gamification!!! I'm enjoying it.

vsalcaidesfaria, 2023-10-27:
I leave the reals that I made for using your code to promote my code club in my school.
https://www.instagram.com/p/Cy5w9tRq9tF/

sgmurray, 2023-10-28:
Thanks for sharing your promo video! It is very exciting to hear that the football game mode might get some use.

There was a bug with the gyro in arena mode that is fixed now. Now, the blue robot in arena mode should chase the ball and, if it successfully grabs the ball, kick the ball into the enemy goal. 

However, the sample code needs 2 tweaks to work with the red robot. 
1. Regardless of the starting orientation, the gyro sensor reads zero. If you want to know the direction the robot is currently facing relative to the field, you need to know the direction the robot was facing at the start and add that to the value the gyro sensor gives you. The blue robot just happened to be facing the 0 direction at the start, so there is no need to add anything in the sample code. However, the red robot is facing the opposite direction (180 degrees).
2. After you fix 1, the red robot will kick the ball into its own goal. You should change the code to point towards the other goal.

You can also use the gps sensor to detect whether the robot is the red robot or the blue robot when the program starts. This way you can have one version of the code that can play either red or blue.